system76/laptop-suggestions #132: Cam, Mic, and Wifi switches

bhillard, 2020-04-17:
I think hardware switches to kill the camera's, microphones, and wireless card would be amazing. These are the only things keeping me from buying a System76 laptop this exact moment.

Dmole, 2020-06-06:
just unplug them and use a rigid cable webcam like this;
https://www.phidgets.com/?tier=3&catid=97&pcid=77&prodid=363

Max689689, 2022-03-11:
> Why/Privacy
> 
> I think hardware switches to kill the camera's, microphones, and wireless card would be amazing. These are the only things keeping me from buying a System76 laptop this exact moment.

I second this.

Hardware switches are a must and are one of the reasons for Frame.Work's success. Not many other brands have them and if they do it's only for one of their models.

hack-char, 2024-08-09:
Hardware switches to kill the webcam, mic and wifi/bluetooth would be most welcome. First thing I did was open the laptop and disconnect them. Thankfully, that was very straight forward and easy for someone familiar with hardware.