system76/laptop-suggestions #133: Ultraportable laptop with 7nm/5nm ARM processor

as4000, 2020-04-28:


Dmole, 2020-06-06:
I've been interested in testing Linux on the 16GB RAM Samsung or Microsoft ARM offerings.

GreyXor, 2020-07-17:
RISC-V please, open ARM alternative

Dmole, 2020-07-17:
> RISC-V

is not actually a product yet, one step at a time.
There is a lot of software to sort out even when fully functional ARM hardware is available;

https://github.com/Sonicadvance1/linux/issues/27

xbjfk, 2021-02-13:
[SiFive](https://www.sifive.com/) make completed RISC-V SoCs, and 95% of Debian's software repository is compatible with RISC-V

Dmole, 2021-02-13:
> [SiFive](https://www.sifive.com/) make completed RISC-V SoCs...

With 95% of of IP on the chip proprietary (ddr,pci,aes,vpu,gpu,etc).

I'll order a BeagleV anyway but there are reasons Apple went with ARM.