system76/laptop-suggestions #134: My laptop requirements

vlada-dudr, 2020-05-10:
### Why/User Benefit/User Problem
The problem is to replace my Thinkpad T520 with anything matching it.

### The "i'm not buying if you don't have" features:
 - comparable keyboard quality to thinkpad (curved keys, long travel, good feedback)
 - trackpoint, only cursor moving interface I use
 - 3 trackpoint buttons (and Tx20 series had the best ergonomics)
 - user serviceable

### Important but no showstoppers:
 - good port options, being future-proof and usable with old equipment also - ethernet, usb a, usb-c, audio jack, each version of usb should be there at least 2x
 - 16:10 or 3:2 screen

### Nice to have
 - leds for power and battery on both sides of the lid
 - no touchpad option (we have trackpoint, remember?)
 - decent headphone amp
 - keyboard available as freestanding usb keyboard (no brain switching when docked in)

### other random things
 -  usage ergonomics is way more then anything else: i don't care for extra 200grams if the keyboard is going to be better, or 5mm more to have proper ethernet port
 - if it is under 2.5kg i'm fine, under 2kg i am happy
 - i have no need for higher then 1920xX resolution
 - amd is cooler then intel
 - i have no need for dGPU
 - i don't care much about battery life
 - frameless looks cool, but is useless in the end
 - what about having a stand for it with adjustable height and click-in dock built-in? 
 - some people are picky about keyboard layout, maybe having "joker" keys, which one can map in system (bios-like) setup? Stuff like printscreen/context menu/fn/caps lock/scroll lock/sys rq/pause...
 - the actual keyboard layout is not important if it is well-thought on well-done hardware, one can learn it

Dmole, 2020-06-06:
trackpoints don't support gestures (zoom/pan/scroll/app-switch/custom/etc).
Between that and the people that only use a keyboard and never the mouse there is a reason trackpoints are not more popular. Whatever ones personal preferences are I think we can understand companies targeting their largest audience.

green7ea, 2021-04-14:
This is my exact feature list. Trackpoint and keyboard are spot on. The keyboard also has to be centered (no numpad).

I'd bump the AMD > Intel a bit more

vlada-dudr, 2021-04-20:
If there is good keyboard with trackpoint, i do not care about Intel vs AMD. The usability issue way more important then processor. It doesn't matter if it has greatest CPU in the world, when the keyboard is not usable.

I do agree with the need of keyboard being centered. It is pretty important to have trackpoint in axis of the laptop.