system76/laptop-suggestions #136: 47W TDP CPU w/o Dedicated Graphics

funnylookinhat, 2020-06-25:
### Why/User Benefit/User Problem
Most developers nowadays prefer portability over gaming on their work machine.  I want to be able to get great battery life and display support without switching between graphics modes in PopOS.

17W TDP CPUs just don't cut it if you're running enough at the same time.  Basically give me the Galago UltraPro.

### Description of the feature

Provide laptops with 47W TDP CPUs that use integrated graphics.

briggs1978, 2020-12-03:
Couldn't agree more as I'm typing this on a Dell Precision 5550 with a 45w CPU and no dedicated GPU.

adeptg, 2022-02-09:
+1 Also, modern integrated graphics are quite powerful on it's own

Aegor1998, 2022-03-09:
Couldn't you just have a mux chip with 2 modes. I CPU heavy made and a GPU heavy mode?