system76/laptop-suggestions #139: Ability to use external monitors without using dedicated GPU

chunga2020, 2020-07-31:
### Why/User Benefit/User Problem

Having extra screen space is very useful. Currently, the only way to use an external monitor is by switching to the dedicated GPU; neither the `intel` nor the `hybrid` options of `system76-power graphics` work. This is not ideal as it means the laptop can get loud sooner as the fans spin up. In Windows, for instance, despite limiting maximum CPU performance to 50% (Windows doesn't seem to like this hardware), I can use my external monitor just fine, and the laptop stays quiet.

### Description of the feature

timnolte, 2020-12-21:
I'm a bit disappointed by this fact and now regret asking my employer to buy me the Oryx Pro since a week be daily connecting and disconnecting an external display but want to use the integrated graphics driver with not connected to maintain decent battery life.

leviport, 2020-12-21:
@timnolte you tried hybrid graphics mode yet?

timnolte, 2020-12-21:
@leviport 
> @timnolte you tried hybrid graphics mode yet? 

From everything I've seen in reviews and posted the hybrid mode only works for the built-in display you are required to reboot into dedicated GPU mode in order to use an external connected display.

leviport, 2020-12-21:
That is old information. Hybrid now works for external displays.

timnolte, 2020-12-22:
@leviport really, interesting, was there a Core Boot update that fixed this in the last few of months? A couple of reviews from like the last few months seemed to still show that that this was a problem. I'm excited to hear that this has been resolved.

leviport, 2020-12-22:
It was the Nvidia 455.xx driver update actually. It brought prime offloading to Linux so that the iGPU can do some rendering and pass it through the dGPU to display ports. It's not without its [growing pains](https://github.com/pop-os/nvidia-graphics-drivers/issues/61) but it was a huge improvement over having to switch to Nvidia graphics mode every time you want to use an external display.

radupopa2010, 2022-12-20:
https://github.com/pop-os/nvidia-graphics-drivers/issues/61#issuecomment-1358664499

This is still happening
```
nvidia-driver-525/jammy,now 525.60.11-1pop0~1670285463~22.04~b24c135 amd64 [installed,automatic]

lsb_release -a
No LSB modules are available.
Distributor ID:	Pop
Description:	Pop!_OS 22.04 LTS
Release:	22.04
Codename:	jammy

lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
01:00.0 VGA compatible controller: NVIDIA Corporation GA104 [Geforce RTX 3070 Ti Laptop GPU] (rev a1)
```

leviport, 2023-06-20:
If you have a bug to report, this is not the repository to do so in. This repo is for suggestions on features for our upcoming in-house laptops. Since external displays work on integrated graphics now, and this issue seems to be confusing users, I'm going to close it.

closed 2023-06-20 by leviport