system76/laptop-suggestions #143: AMD Cezanne (5xxx) (Zen 3)

GreyXor, 2021-01-14:
### Why/User Benefit/User Problem
Notebooks with AMD 5000 series processors APU (Cezanne, zen 3)

### Description of the feature
Great energy savings, great power. Much better than their Intel equivalent.
https://en.wikichip.org/wiki/amd/microarchitectures/zen_3

Dmole, 2021-03-18:
5 looks the same as 4 with a slight clock boost, and you can get 4 now.

GreyXor, 2021-03-29:
> 5 looks the same as 4 with a slight clock boost, and you can get 4 now.

Absolutely not, Zen 3 Cezanne is lot more than just a clock boost

Dmole, 2021-03-29:
> Absolutely not, Zen 3 Cezanne is lot more than just a clock boost.

https://en.m.wikipedia.org/wiki/Zen_3

Reads to me like a less secure cache, and increased latency vs a bit of speed. No security fixes or new user facing features.