system76/laptop-suggestions #145: Modular laptops

anubhavkini, 2021-05-17:
### Why/User Benefit/User Problem
Easy repairs, upgrades and customization.
### Description of the feature
There's a new company called [Framework](https://frame.work/) that's releasing a modular laptop. I think it's a great idea and maybe System76 would be interested.

TeamLinux01, 2021-07-27:
Now that the framework is starting to ship, it might be possible to make compatible and interchangeable parts for S76 and framework machines, if S76 wants to join in that front.

I personally would enjoy a modular laptop that is repairable/upgradable with open firmware and EC.

adeptg, 2021-11-10:
It looks like the Framework laptop [works well enough with Linux](https://www.phoronix.com/scan.php?page=article&item=framework-laptop&num=1). Having an OSS firmware and PopOS on the Framework would be nice.

curiousercreative, 2022-07-15:
Just partner with Framework. They've already figured out hardware design and manufacturing.

adeptg, 2022-07-16:
@curiousercreative   agree. But this way is so obvious that I think, they have already tried. And, as we don't see results yet, means that, for now, it doesn't possible for some reason.
Personally, I would buy a Framework laptop from System76 with a little premium for OS and firmware support