system76/laptop-suggestions #152: An ideal laptop for artists - Stylus w/ Pressure Support and Programmable Buttons

KameiKojirou, 2022-07-08:
I'm an artist that uses Pop!_OS professionally in my business (Blender/Krita/Inkscape). I would love to buy a System76 laptop, but I don't because they do not have stylus pressure support as far as I can tell. By that I mean specifically the built in display. I would love a laptop designed with this and even better a section of well laid out programmable buttons that could be mapped to keyboard/mouse shortcuts potentially on the back of the display lid. I'd suspect if it had CPU/GPU/RAM many other artists be all over it, including myself. In my opinion at least almost all of the options out there currently are lacking in one area or another. Anyways, that'd be the Bees Knees.

SherlockHomie, 2022-10-15:
+1 Remote/hybrid async business work requires lots of markup to screenshots, whiteboarding, etc even for non-creative professionals :) 

It is a lot easier for me to explain an issue or architecture by marking up a screenshot or gif/video (yes drawing on video/gif recordings is super easy in linux, see: https://extensions.gnome.org/extension/4937/draw-on-you-screen-2/). I can cut down on several back+forth async communication with the help of a pen.

My target is a ~2.5lb 2in1 laptop that fully supports linux, and I believe the only one that meets that is the Lenovo X13 line (https://ubuntu.com/certified/laptops?q=x13&limit=20&vendor=Lenovo) so that's what I'll likely grab for now. Would love to support System76 in the future!

sidera-dev, 2025-05-20:
Please make it!|. I’m wating from a life a Linux 2in1 Laptop / tablet with decent performance and wacom aes and native support.

doctorjei, 2025-07-01:
I am 100% in favor of a 2-in-1. If you build it, please use the "double hinge" approach of the Surface Studio laptop or Acer ConceptD Ezel, and not the crummy "now my keyboard is the back of my laptop" kludge used by most. 😅

Just one data point, but I love Pop, and I'm I'm the market for a new laptop soon. I need a 2-in-1, but as much as it pains me, there's no System76 model for me at the moment. :(